qeda/qeda #1: Переменный резистор

scuratovo, 2016-08-16:
_Добрый день!
Огромное спасибо за вашу работу, это именно то, что, по-моему личному мнению, все сообщество ждало многие годы!_

Попробовав ваше приложение я столкнулся с нехваткой генерации переменного резистора. Я далеко не программист, но попытался добавить этот функционал в файл ..\qeda\lib\symbol\default\resistor.js

  [image]
Вышло сносно, можно ли Вас попросить добавить этот функционал в мастер ветку?

Файл прилагаю (прошу прощения за такой способ, я еще не освоил работу с github)
[resistor.js.txt](https://github.com/qeda/qeda/files/420641/resistor.js.txt)

_С пребольшим Уважением к Вам и Вашей работе в сфере открытого ПО._

shamilsan, 2016-08-17:
Thanks a lot for your feedback!

There is some issue with adding your code to master branch due to:
- source code is written on CoffeeScript, `lib` subdirectory is fully generated
- resistor should be refactored and generalized for multi-pin purposes as e.g. [diode](https://github.com/qeda/qeda/blob/master/src/symbol/default/diode.coffee) has been

But I promise to add these options as soon as possible.

scuratovo, 2016-08-17:
Okay,I'l  think about it.