qeda/qeda #2: suffix in housing

steel-ne, 2016-09-10:
Ругается, если в housing есть suffix, как в ti/lm158:

di@ubuntu:~/Workspace/pa-01/schemes$ qeda --version
QEDA v0.2.0
di@ubuntu:~/Workspace/pa-01/schemes$ qeda generate wa
Read 'capacitor/c0603'
Read 'capacitor/c1210'
Read 'capacitor/c0805'
Read 'resistor/r0805'
Read 'misc/tp-th-1'
Read 'st/stm32f030c'
Read 'ti/lm158'
/usr/local/lib/node_modules/qeda/lib/qeda-library.js:110
        suffixes = obj.housing.suffix.replace(/\s+/g, '').split(',');
                                     ^

TypeError: Cannot read property 'replace' of undefined
    at QedaLibrary.add (/usr/local/lib/node_modules/qeda/lib/qeda-library.js:110:38)
    at generateLibrary (/usr/local/lib/node_modules/qeda/bin/qeda:155:9)
    at Object.<anonymous> (/usr/local/lib/node_modules/qeda/bin/qeda:302:5)
    at Module._compile (module.js:410:26)
    at Object.Module._extensions..js (module.js:417:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Function.Module.runMain (module.js:442:10)
    at startup (node.js:136:18)
    at node.js:966:3

steel-ne, 2016-09-10:
версия  0.2.1 - ошибка сохраняется

shamilsan, 2016-09-10:
Unfortunately there is some incompatibility between v0.1 and v0.2 component YAML-descriptions (these compatibility issues are possible until project reaches the v1.0 milestone).

Please try to re-download component descriptions from remote repository (using `qeda load` command or by deleting old ones and repeating `qeda generate` command).  

If problem persists could you please attach your YAML-description to this issue.

steel-ne, 2016-09-10:
yes, I've missed library convertion to v0.2, so I made own components based on v0.1 specs. I'll repare them and send pull request soon

closed 2016-09-10 by steel-ne