qeda/qeda #3: symbol generation

steel-ne, 2016-09-12:
Attached test.yaml expected to generate 8-pin IC symbol: left side pins 1-4, right side others
[test.yaml.txt](https://github.com/qeda/qeda/files/466509/test.yaml.txt)
[image]

shamilsan, 2016-09-12:
Thank you for information. There was a bug indeed. Now this bug is fixed (see v0.2.3).

But note that `PF2-3` and `PC2-3` are not grouped together because they are not groups but pin lists.

My opinion that it is better to group pins by their primary function. Alternative pin functions are not needed to be combined in groups.

```yaml
pinout:
  # Primary pin functions
  PORTC:
    PC0-3: 1, 2, 7, 8
  PORTF:
    PF0-3: 3-6

  # Alternative pin functions
  OSC_IN: 1
  OSC_OUT: 2
  OSC32_IN: 3
  OSC32_OUT: 4

schematic:
  symbol: IC
  left: PORTC
  right: PORTF
```
This will produce following symbol:
[image]

steel-ne, 2016-09-12:
Thank you for fix!
I think, OSC - is the main function on this pins :)

And another question - is there a simple way to distribute pins in symbol by their numbers, to look like a footprint? Sometimes, when drawing a scheme with large gpio buses (matrix keyboard, leds, lot of relays) I choose adjanced pins to simplify routing. And such symbol can help to choose pins wisely, without switching to pcbEditor and back

shamilsan, 2016-09-12:
Unfortunately there is no such option. But it is a good idea, I will think about this functionality.

closed 2016-09-12 by shamilsan