qeda/qeda #5: Additional links (Feature request)

Gasman2014, 2016-09-19:
Would there be any interest in adding an optional SPICE model within the library or an option to link to an external SPICE model (now that KiCAD is offering built-in simulation...)? 

Similarly, any plan to provide links to .STEP models for MCAD?

shamilsan, 2016-09-19:
There is no problem to add link to SPICE model, but that is a question how to use it because I did not use SPICE modeling in KiCad. I will investigate this option, thanks for idea.

Regarding adding 3D-models, this work is scheduled. 3D models are planned to be generated automatically for standard packages. As far as I know supporting of STEP models instead of VRML is added to KiCad dev-branch only. My plan is to add 3D models support after QEDA become cloud service (this is a next big milestone planned).

Gasman2014, 2016-09-19:
I think the SPICE modelling is currently still in dev but it was announced as in the roadmap a couple of months back and I think it is a compiled option from source - see https://youtu.be/A2_-hdRcf4U . Presume it is just Eeschema  -> SPICE netlist. Would have though that being able to hook into a SPICE model embedded in the component itself might be quite a good feature. SPICE models are also quite small text files so should not be too storage dependant.

Regarding 3dStep models; I think you are right - native .step support is also still in dev but you can use .step models with the help of a couple of macros in FreeCad - have a quick look at https://forum.kicad.info/t/kicad-stepup-new-exporter-for-3d-mcad-feedbacks-are-welcome/1048/38
The requirements are that there are matched .wrl/.stp pairs of identically named models e.g. C_1206.wrl and C_1206.stp and there is a bit of adjustment necessary to ensure the scale and placement correspond. Interesting links there also to the parametric housing builder script for various style packages. Think this is something that could easily be incorporated but .step models can be big!!

Library management in KiCad is desperately needing something like this.

shamilsan, 2016-09-20:
Interesting information, thanks a lot. Will examine it.