qeda/qeda #14: Adopting new IPC-7351C guidelines

shamilsan, 2016-10-10:
Changes are to be supported:
- Land pattern naming conventions
- New pad calculation rules
- Silkscreen updates
- Terminal leads to mechanical layer

Gasman2014, 2017-02-27:
Will this enhancement also address the orientation of the generated footprints? The zero orientation (IPC7351 A or IEC 61188-7 B) should be selectable. Also the zero orientation should be horizontal for two pad components with pad 1 to the left. The generated footprints are presently vertical for two pad components and zero aligned step/vrml models need to be rotated to fit.

Gasman2014, 2017-03-13:
[image]

Zero component orientation should be horizontal for two pin components in IPC7351A. Current output is vertical and hence components require rotation.

QEDA
[image]
[image]

 IPC7351
[image]
[image]

syakupov, 2017-03-18:
Yes you are right. This is due to `bodyWidth` and `bodyLength` confusion. Now `bodyWidth` is always horizontal. Now I understand this was bad decision. I hope this issue will be fixed in v0.4. But unfortunately new version will be incompatible with old libraries.

I will think about customizable zero position in new `ipc-7351c` style.

shamilsan, 2017-03-18:
Yes you are right. This is due to `bodyWidth` and `bodyLength` confusion. Now `bodyWidth` is always horizontal. Now I understand this was bad decision. I hope this issue will be fixed in v0.4. But unfortunately new version will be incompatible with old libraries.

I will think about customizable zero position in new `ipc-7351c` style.

Gasman2014, 2017-03-19:
I don’t think backward compatibility is a big issue. You will not be reading the libraries back into QEDA and the .yaml files don’t need to change. You just need to add the ‘IPC_Zero position’ setting as an option (i.e. maintain backward compatibility OR adopt IPC zero rotation) and then generate the footprints horizontally not vertically. Think this is only an issue for 2-part components in QEDA anyhow? The whole rotation thing is quite confusing esp the EiA481 / vs IPC.

John

dvc94ch, 2018-01-26:
Any advice on how to get a copy of ipc-7351c? I can't find a purchase link anywhere... Has it even been released yet officially?

Gasman2014, 2018-01-26:
Aparantly progress seems to have stalled. https://www.pcblibraries.com/forum/ipc7351c-draft-or-release-date_topic1818.html

According to this, the amended solder joint goals in the documents section have been updated to include all the agreed changes so should be c compliant. 

You have to register but it is free.

Houkime, 2019-08-11:
So... if IPC is effectively dead...
It seems like oshw community should produce guidelines of their own now and compile stuff from PCB manufacturers, datasheets and experience?

Is there a repo for that already?