qeda/qeda #18: generation empty ALIAS

steel-ne, 2016-10-17:
lines with empty ALIAS property in .lib files generate errors in kicad for windows

for example command `qeda test capacitor/c0603 `
generates such library:
```
EESchema-LIBRARY Version 2.3 Date: 17/10/2016 15:35:34
#encoding utf-8
#
# C0603
#
DEF C0603 C 0 40 N N 1 L N
F0 "C" 0 110 50 H V C BNN
F1 "C0603" 0 -110 50 H V C TNN
F2 "test_c0603:CAPC1608X92M" 0 0 0 H I C CNN
ALIAS 
$FPLIST
  CAPC1608X92M
$ENDFPLIST
DRAW
X 1 1 -100 0 85 R 50 50 1 1 P
X 2 2 100 0 85 L 50 50 1 1 P
P 2 1 1 12 -15 80 -15 -80 N
P 2 1 1 12 15 80 15 -80 N
ENDDRAW
ENDDEF
# End Library
```

[image]

removing empty ALIAS line cancels this error

qeda 0.2.6
kicad:
```
Application: kicad
Version: (2016-09-17 revision 679eef1)-makepkg, release build
Libraries: wxWidgets 3.0.2
           libcurl/7.46.0 OpenSSL/1.0.2d zlib/1.2.8 libidn/1.32 libssh2/1.6.0 librtmp/2.3
Platform: Windows 7 (build 7600), 64-bit edition, 64 bit, Little endian, wxMSW
- Build Info -
wxWidgets: 3.0.2 (wchar_t,wx containers,compatible with 2.8)
Boost: 1.57.0
Curl: 7.46.0
KiCad - Compiler: GCC 5.2.0 with C++ ABI 1009
        Settings: USE_WX_GRAPHICS_CONTEXT=OFF
                  USE_WX_OVERLAY=OFF
                  KICAD_SCRIPTING=ON
                  KICAD_SCRIPTING_MODULES=ON
                  KICAD_SCRIPTING_WXPYTHON=ON
                  BUILD_GITHUB_PLUGIN=ON
                  KICAD_USE_SCH_IO_MANAGER=ON
                  KICAD_USE_OCE=ON

```

steel-ne, 2016-10-17:
in Kicad file formats pdf:

> 2.3.1 - Description of Aliases
> This line exists **only** if the component has alias names.
> format:
> ALIAS name1 name2 name3…

shamilsan, 2016-10-17:
Thank you for info. Fixed in v0.2.7.

closed 2016-10-19 by steel-ne