qeda/qeda #22: duplicated pins at schematic symbol generation

tiborgats, 2017-02-01:
When schematic symbol is generated to the attached part, many pins are duplicated. Another problem is that in this case some pins are placed outside of the IC symbol's rectangle.
A workaround could be that I only use a signal name once, but that would not be correct. According to the part's [datasheet](http://www.nxp.com/assets/documents/data/en/data-sheets/KL02P20M48SF0.pdf) (page 35), the same signals can be accessed through multiple pins (with software configuration).
Anyways, if something is not correct in the .yaml file, an error message would be nicer than generating a messed-up part...
[mkl02z32caf4r.yaml.zip](https://github.com/qeda/qeda/files/744282/mkl02z32caf4r.yaml.zip)

I would really appreciate if you can give a quick solution, because I have a very tight deadline on this project. Thanks!
Btw, I love the concept of QEDA!

shamilsan, 2017-02-01:
When you specify `schematic.left` and `schematic.right` section you are to use groups of unique pins (i.e. each pin number should be included only once in group and pin numbers cannot be shared between groups). 

You are right about error message requirement. There is a lack of diagnostics in QEDA at the moment and I will improve it in future.

Thank you for feedback!

Please see attached corrected yaml.

[mkl02z32caf4r.zip](https://github.com/qeda/qeda/files/744623/mkl02z32caf4r.zip)

[image]

tiborgats, 2017-02-01:
Wow, that was really quick!  Thank you very much for your support!

closed 2017-02-25 by shamilsan