qeda/qeda #23: 3d models

Gasman2014, 2017-02-23:
No 3d models being generated yet. Is this still in the planning stage?

shamilsan, 2017-02-24:
There is experimental 3D support added recently (not included to release yet). Not realistic models but body boxes.
[image]

tiborgats, 2017-02-24:
If it would have body cylinders also, it will be already very useful for checking mechanical designs.

Gasman2014, 2017-02-24:
For mechanical models, a STEP model would be preferable. I think the present output is .wrl.

tiborgats, 2017-02-24:
True, `.STEP` is more universal, and already handled by the nightly version of KiCad. If it helps the development of QEDA, some nice `.STEP` examples can be generated by the evaluation version of [PCB Library Expert](http://www.pcblibraries.com/Products/FPX/KiCad.asp).

shamilsan, 2017-02-24:
Looking forward to STEP support to be added to main release of KiCad. But only VRML is supported at the moment.

Gasman2014, 2017-02-24:
Yes but STEP has been supported directly for some while in the nightly builds

> Note
> UPDATE 2016-09-07 The OCE plugin for STEP and IGES 3D viewer support was merged with the offical development branch recently and should now be available in the newest nightlies.

I don't imagine that it will be too much longer before STEP support reaches the stable release. 4.0.6 has just been released and I have seen some comments on the developer site hinting at a spring release for a 4.5 version and this would include a cherry picked selection of the stuff presently in the nightly builds. The v5 roadmap as recently outlined by Wayne S (https://drive.google.com/file/d/0B8ng6CtJ1Hh6NU11UTRrbzZBYjg/view) certainly plans to include more 3d model support and I am fairly sure that the current OCE STEP support has been considered stable.

tiborgats, 2017-09-10:
It would be nice to have an option for the footprint: a path to an already existing 3D file. (And if it is given, you don't have to generate a 3D model for that footprint.) Many complicated parts have already existing 3D models downloadable from their manufacturers in STEP file format, I would prefer to use them.

m-byte, 2023-07-28:
This might be a few years late, but at least for KiCad7, custom models are possible since https://github.com/qeda/qeda/pull/102. If anyone needs support for older versions, of KiCad, please open a new issue. VRML models have also been support for years now, so I'm closing this.

closed 2023-07-28 by m-byte