qeda/qeda #25: Rounded rectangle pad shape option

tiborgats, 2017-02-24:
There is a new pad shape option in KiCad (in the nightly build version): Rounded Rectangle.
It is the preferred pad shape in the IPC-7351C standard, so it would be useful if QEDA had this option under `housing`.
- the default corner radius is 25% of the `leadWidth`
- the default maximum radius is 0.25mm

A pad example with this property looks like this in the `.kicad_mod` file:
`(pad 1 smd roundrect (at 1.5 1.1) (size 2 1.524) (layers F.Cu F.Paste F.Mask)(roundrect_rratio 0.25))`

shamilsan, 2017-02-24:
I am waiting for this KiCad's feature to be added to main release.
Thanks for reminder, I will schedule this task.

tiborgats, 2017-02-24:
I propose to have a `leadShape` option under `housing` for this. The values would be the followings:
- `Circular` or `C`
- `Oval` or `O`
- `Rectangular` or `R`
- `Trapezoidal` or `T`
- `RoundedRectangle` or `RR`
- and if possible, for D shape: `D`

tiborgats, 2017-02-24:
I think you don't have to wait for this feature to be added to the main KiCad release. It can simply be optional in QEDA. Those who use the release version of KiCad, they can skip this option, but many people use the daily build of KiCad, for them it would already be useful. In the documentation of this feature you can put a warning about it. Or as an alternative solution, you can put an option in QEDA to choose the target KiCad version.

closed 2017-02-25 by shamilsan