qeda/qeda #26: padHeight

tiborgats, 2017-02-24:
For most of the housings there are `padLength` and `padWidth` parameters. But if I choose `pattern: custom`, instead of `padLength` there is `padHeight`. I think, it is a mistake.
There is also another error in https://github.com/qeda/qeda/blob/master/src/pattern/default/custom.coffee
in line 14:
`padDiameter` is missing from this condition.

shamilsan, 2017-02-24:
There is no `padLength`. 3D objects (lead, body etc.) have `width`, `length` and `height`. 2D objects have `width` and `height` only.
Tab is considered to be 3D so it has `width`, `length`. For some packages (e.g. D2PAK) it has `height` too.

tiborgats, 2017-02-24:
Oh, I see, you are right.

But the other issue I reported about `padDiameter`: in the case I choose `pattern: custom`, and I want to define pads by diameter, it does not work now. In the script it is missing from line 14.