qeda/qeda #29: Tab corner cut for pin 1

tiborgats, 2017-02-25:
Many parts (QFN, SON) often has a Tab with a 45° cut corner for signing pin 1 position. It would be nice to be able to define a `tabCut` parameter, which tells the cut distance from the corner.

shamilsan, 2017-02-25:
I like this idea but as far as I know there is a lack of polygonal pads in KiCad. I do not like multiple pads with same numbers forming complex shape due to mask issues.

tiborgats, 2017-02-25:
You are right, complex shape with multiple pads would be an ugly solution. I searched for possibilities, and found this:
https://bugs.launchpad.net/kicad/+bug/1660151
So, it will be possible soon, but it is not included in the nightly builds yet.

I think it would be a good policy to have features like this in QEDA after it will be available with KiCad nightly builds, and don't wait till it lands in the stable KiCad version. If it is optional, it will not bother those who use the stable KiCad, but could be useful for the ones use the nightly. And also helps the earlier discovery of possible bugs in new KiCad features.