qeda/qeda #34: FET symbol broken

tiborgats, 2017-08-10:
I tried to create a BGA housed FET in QEDA, but I get a broken schematic symbol:
[image]

It would be nice to have `G`, `D`, `S` signs on the legs instead of numbering, as for the BGA component numbering is ugly and not useful at all. Also please allow thermal (not connected) pads.

I made a custom housing as the BGA pattern does not allow custom pad numbering scheme.

The `epc2047.yaml` is this:
```
name: EPC2047
description: eGaN FET, 200V, 10mΩ
datasheet: http://epc-co.com/epc/Portals/0/epc/documents/datasheets/EPC2047_preliminary.pdf
keywords: eGaN, GaN, FET

pinout:
  G: 1
  D: 3, 4, 5, 9, 10, 11
  S: 2, 6, 7, 8, 12, 14
  substrate: 13

properties:
  in: G
  passive: S, D
  nc: substrate

schematic:
  symbol: FET
  options: n, diode

housing:
#  pattern: BGA # WLCSP
#  leadCount: 14 # 14
#  rowCount: 3
#  columnCount: 5
#  horizontalPitch: 1.0
#  verticalPitch: 0.5
#  leadDiameter: 0.330
  pattern: custom
  bodyLength: 1.570-1.630
  bodyWidth: 4.570-4.630
  height: 0.790
  padDiameter: 0.330
  padWidth: 0.330 # workaround for QEDA bug, can be deleted later
  padHeight: 0.330 # workaround for QEDA bug, can be deleted later
  padPosition: -2, -0.5, -2, 0.5, -1, -0.5, -1, 0, -1, 0.5, 0, -0.5, 0, 0, 0, 0.5, 1, -0.5, 1, 0, 1, 0.5, 2, -0.5, 2, 0, 2, 0.5
```

tiborgats, 2017-08-16:
It would be nice if QEDA could generate something like this:
[image]
```
EESchema-LIBRARY Version 2.3
#encoding utf-8
#
# EPC2047
#
DEF EPC2047 VT 0 0 N Y 1 F N
F0 "VT" -200 100 50 H V R BNN
F1 "EPC2047" 0 250 50 H V R TNN
F2 "RatBand_nominal:EPC_EPC2047" 0 0 0 H I C CNN
F3 "http://epc-co.com/epc/Portals/0/epc/documents/datasheets/EPC2047_preliminary.pdf" 0 0 0 H I C CNN
$FPLIST
 EPC_EPC2047
$ENDFPLIST
DRAW
C 0 0 150 1 1 12 f
P 2 1 1 12 -150 -100 -67 -100 N
P 2 1 1 12 -67 100 -67 -100 N
P 2 1 1 12 -37 -74 150 -74 N
P 2 1 1 12 -37 -48 -37 -100 N
P 2 1 1 12 -37 0 38 0 N
P 2 1 1 12 -37 26 -37 -26 N
P 2 1 1 12 -37 74 150 74 N
P 2 1 1 12 -37 100 -37 48 N
P 2 1 1 12 38 0 38 -74 N
P 2 1 1 12 68 16 105 16 N
P 2 1 1 12 86 -16 86 -74 N
P 2 1 1 12 86 74 86 16 N
P 2 1 1 12 150 -74 150 -100 N
P 2 1 1 12 150 74 150 100 N
P 3 1 1 12 0 15 -37 0 0 -15 F
P 4 1 1 12 68 -16 105 -16 86 16 68 -16 N
X G 1 -350 -100 200 R 50 50 1 1 I
X S 2 150 -300 200 U 50 50 1 1 P
X D 3 150 300 200 D 50 50 1 1 P
X D 4 150 300 200 D 50 50 1 1 P
X D 5 150 300 200 D 50 50 1 1 P
X S 6 150 -300 200 U 50 50 1 1 P
X S 7 150 -300 200 U 50 50 1 1 P
X S 8 150 -300 200 U 50 50 1 1 P
X D 9 150 300 200 D 50 50 1 1 P
X D 10 150 300 200 D 50 50 1 1 P
X D 11 150 300 200 D 50 50 1 1 P
X S 12 150 -300 200 U 50 50 1 1 P
X sb 13 0 -300 150 U 50 50 1 1 P
X S 14 150 -300 200 U 50 50 1 1 P
ENDDRAW
ENDDEF
#
#End Library
```

shamilsan, 2017-09-12:
This is the bug indeed, thank you for reporting. About adding the case pin I need to investigate.