qeda/qeda #43: decaffeinate

dvc94ch, 2018-01-26:
Hi!

Since es6 there isn't really much reason imo to continue using coffescript. Migrating to es6 would potentially increase the developer base.

Would a PR that decaffeinates the code base have any chance at being merged?

Thanks,
David

[0] https://github.com/decaffeinate/decaffeinate

shamilsan, 2018-01-26:
Hi,

I am not sure about new JavaScript, sorry. Despite ES6 it is still quite ugly. Personally I hate this
```
if (typeof elvis !== 'undefined' && elvis !== null)
```
instead of
```
if elvis?
```
JavaScript itself is a bad programming language. But it is needed to be used in QEDA project due to future embedding to browser code (online component editor). Therefore I am forced to use its dialect. CoffeeScript looks more appropriate at the moment.

Another question is that some code refactoring must be done.  My opinion this task is more important.

dvc94ch, 2018-01-27:
This can be written as `if (elvis?)` using the babel transpiler and has reached stage 2 in the standardization process for ecma script 2018.

[0] https://github.com/tc39/proposal-optional-chaining
[1] https://github.com/babel/babel/issues/855

dvc94ch, 2018-01-27:
Anyway just to test what the generated code would look like, I ran decaffeinate on the code base. Noticed one issue with this being used before super(). Since every Icon needs a height and a width, it should be supplied as constructor arguments, which is in my opinion a good practice anyway.

```js
class Icon {
  constructor({symbol, element, width, height}) {
    this.symbol = symbol;
    this.element = element;
    this.schematic = this.element.schematic;
    this.settings = this.symbol.settings;
    this.lineWidth = this.settings.lineWidth.thick;
    this.width = width * this.settings.factor;
    this.height = height * this.settings.factor;
  }
}
```

dvc94ch, 2018-01-28:
closing for now. since there is a clear update path to js if coffescript is ever deprecated this can be done when that happens

closed 2018-01-28 by dvc94ch