qeda/qeda #44: svg backend

dvc94ch, 2018-01-26:
A svg backend would be really useful. This would add support for generating schematics using netlistsvg.

[0] https://github.com/nturley/netlistsvg
[1] https://raw.githubusercontent.com/nturley/netlistsvg/master/lib/analog.svg

shamilsan, 2018-01-26:
This is a good idea, I was thinking about it.

There is a need to add `custom` symbol generator. And it should support SVG paths.

dvc94ch, 2018-01-28:
Status update on exporting symbols to svg

- The pin spacing for kicad pins doesn't seem consistent. (resistor pin to rectangle space seems less than transistor-npn pin to circle). I need to find a way to remove it.
- Need to add a way to get the symbol width / height
- Some symbols (nfet and zener) seem to be off (zener additional pad)
- Find a way to add allowed orientations to yaml configuration (horizontal, vertical, horizontal-flipped, vertical-flipped) and allow exporting the symbol in all allowed orientations

[image]

PR #47 (2018-02-02): wip: SVG generator
TODO:
- [ ] improve text positioning
- [ ] handle multi part symbols
- [x] cleanup
- [x] symbols in multiple orientations
- [x] draw builtin symbols / missing symbols in qeda
- [x] add support for adding references/names in netlistsvg
- [x] recognize s:position in netlistsvg and use it to tag lateral ports https://github.com/nturley/netlistsvg/issues/34 https://github.com/nturley/netlistsvg/issues/30

Closes #44 

Symbols:
[image]

Netlistsvg generated schematic:
[image]

Pattern:
[image]
closed 2018-02-02 by shamilsan