qeda/qeda #45: support datasheet parameters in component format

dvc94ch, 2018-01-26:
For matching components using electro-grammar it would be useful to add datasheet parameters like vf, ic, beta, bwp, sr to component definitions.

[0] https://github.com/monostable/electro-grammar

shamilsan, 2018-01-26:
I am sorry but I do not understand what kind of data should be added to component description.

Could you please explain with more details and examples. Following the link above I see only the element values parser. Thank you.

dvc94ch, 2018-01-26:
there's an antlr branch kasbah and I have been working on, it's not in master yet. I'd like to make use of your component library and extend it with some metadata. With the new version of electrogrammar you'll be able to do something like this:

```python
import electrogrammar as eg
components = eg.matchQeda(eg.parse('npn sot23 beta=200 ic>=200mA'))
```

dvc94ch, 2018-08-10:
Don't think this is a good idea anymore. Closing.

FYI: I wrapped `qeda` for use with `electron` the successor of `pycircuit`.
- [0] https://github.com/electron-lang/electron-qeda
- [1] https://github.com/electron-lang/quickstart/blob/master/make-layout.js

closed 2018-08-10 by dvc94ch